adafruit/Adafruit-Motor-HAT-Python-Library #4: Problems with Microstepping

AarBee1, 2015-10-07:
In Adafruit_MotorHAT module:
__________________________
      if (stepstyle == Adafruit_MotorHAT.MICROSTEP):
         # this is an edge case, if we are in between full steps, lets just keep going
         # so we end on a full step
         while (lateststep != 0) and (lateststep != self.MICROSTEPS):
            lateststep = self.oneStep(dir, stepstyle)
            time.sleep(s_per_s)
__________________________

dir refers to a built-in function:  in this scope the correct variable name is direction. Once this change is made, the code works as it should.

Due to the bug: moving 2 steps back and 2 steps forward causes significant drift in every cycle - observable to the eye within 3-4 cycles.

AarBee1, 2015-11-14:
I was trying to figure out the source of inordinate variability in timing of motor motion - just one motor (so not a threading issue), moving slow ~ 0.5rpm (so not an I2C communication rate issue), two step fwd and two step back, with microstepping. Also, there variability would depend on whether I moved fwd first or back first.

It seems like the line towards the end of  the step function
`while (lateststep != 0) and (lateststep != self.MICROSTEPS):`
should actually read
`while (lateststep != 0) and (lateststep != self.MICROSTEPS*4):`

I am not a 100% sure of this as I don't understand the oneStep code fully.

marekpechal, 2018-08-02:
There does seem to be an issue with the line
`while (lateststep != 0) and (lateststep != self.MICROSTEPS):`
I found this when I noticed that running
`for _ in range(400): stepper1.step(1,Adafruit_MotorHAT.FORWARD,Adafruit_MotorHAT.MICROSTEP)`
actually results in twice as many steps as
`stepper1.step(400,Adafruit_MotorHAT.FORWARD,Adafruit_MotorHAT.MICROSTEP)`

I think the line in question should read (at least it fixed the problem for me):
`while (lateststep != 0) and (lateststep != self.MICROSTEPS) and (lateststep != 2*self.MICROSTEPS) and (lateststep != 3*self.MICROSTEPS):`

caternuson, 2020-10-02:
This library is being deprecated. Please update to using the new library here:
https://github.com/adafruit/Adafruit_CircuitPython_MotorKit
If you continue to have this issue, please open a new issue in that repo.

closed 2020-10-02 by caternuson