adafruit/Adafruit-Motor-HAT-Python-Library #9: performance + other improvements

Zimcoding, 2016-06-02:
Hello,

I started to use the Adafruit motor hat with this library.
I discovered several issues:

1.	When a new motorhat is initialised, a new steppermotor instance is also initialized in line 230:
`self.steppers = [ Adafruit_StepperMotor(self, 1), Adafruit_StepperMotor(self, 2) ]`

In this way. The steps per revolution is not changeable during initialization and remains 200.

2.	The timing is of by using time.sleep(s_per_s) in line 159. This can be greatly Improved by:
`time.sleep(time.time() - time_before_start - s_per_s)`

3.	The speed of execution can be improved by only sending the commands needed with this snippet starting from line 130:
```
           if (dir == Adafruit_MotorHAT.FORWARD):
                   prev_coil = step2coils[(self.currentstep/(self.MICROSTEPS/2) - 1) % len(step2coils)]
           else:
                   prev_coil = step2coils[(self.currentstep/(self.MICROSTEPS/2) + 1) % len(step2coils)]
           coils = step2coils[self.currentstep/(self.MICROSTEPS/2)]

#print "coils state = " + str(coils)
pin_to_set = [self.AIN2, self.BIN1, self.AIN1, self.BIN2]
for pin in range (0, 4):
           if prev_coil[pin] != coils[pin]:
                      self.MC.setPin(pin_to_set[pin], coils[pin])
```

4.	Is it possible that at line 239, the logic bits are reversed? 

```
if (value == 0):
       self._pwm.setPWM(pin, 0, 4096)
if (value == 1):
       self._pwm.setPWM(pin, 4096, 0)
```

Shouldn’t it be like this:
```
if (value == 1):
       self._pwm.setPWM(pin, 0, 4096)
if (value == 0):
       self._pwm.setPWM(pin, 0, 0)`
```
5. The loop running the commands is locked and cannot be aborted. This is often needed in all sorts of situations. Is it possible to have an abort bit programmed in the loop in line 157

hendrikmaus, 2016-06-25:
Hey there, I am very interested in your improvements. Would you mind to create a working fork with your suggestions? I'd be happy to give them a try and to provide feedback.

Maybe these changes can help you as well: https://github.com/adafruit/Adafruit-Motor-HAT-Python-Library/pull/10 they did a great improvement in my setup.

Martronic-SA, 2017-04-25:
> time.sleep(time.time() - time_before_start - s_per_s)
Greatly improves the precision of the timing.
Be careful however that the result of this opration might be negative so:
towait = time.sleep(time.time() - time_before_start - s_per_s)
if towait < 0:
    continue
else:
    time.sleep(towait)

caternuson, 2020-10-02:
This library is being deprecated. Please update to using the new library here:
https://github.com/adafruit/Adafruit_CircuitPython_MotorKit
If you continue to have this issue, please open a new issue in that repo.

closed 2020-10-02 by caternuson