adafruit/Adafruit-Motor-HAT-Python-Library #16: StepperTest.py

RobbieRobot23, 2016-11-30:
I've been trying to use the HAT and followed the instructions on Adafruit to a T, but I am still getting the same error message.

pi@raspberrypi:~/Adafruit-Motor-HAT-Python-Library/examples $ sudo python StepperTest.py
Traceback (most recent call last):
  File "StepperTest.py", line 3, in <module>
    from Adafruit_MotorHAT import Adafruit_MotorHAT, Adafruit_DCMotor, Adafruit_StepperMotor
  File "build/bdist.linux-armv7l/egg/Adafruit_MotorHAT/__init__.py", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_MotorHAT-1.3.1-py2.7.egg/Adafruit_MotorHAT/Adafruit_MotorHAT.py", line 52
    if (dir == Adafruit_MotorHAT.FORWARD):
     ^
IndentationError: expected an indented block

I have tested my i2C and everything seems to be working there.

pi@raspberrypi:~ $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: 60 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: 70 -- -- -- -- -- -- --                         

My stepper motor is receiving power because I cannot manually turn it.

Any solution? Thanks in advance.

mllobere, 2016-12-01:
You have an "IndentationError", python expects your code to be indented. https://www.python.org/dev/peps/pep-0008/#indentation 
Post your code if you need more help.

RobbieRobot23, 2016-12-01:
Thanks for the comment.

I'm pretty new to python RasPi and was following the instructions that came with the HAT. With that said I downloaded the software as instructed, which reached to this github account. I followed the path to the 

"/usr/local/lib/python2.7/dist-packages/Adafruit_MotorHAT-1.3.1-py2.7.egg/Adafruit_MotorHAT/Adafruit_MotorHAT.py", line 52

But hit a dead end when I got to zip file (.egg file) "Adafruit_MotorHAT-1.3.1-py2.7.egg" . I tried to unzip it so I could edit it, but no luck.

RobbieRobot23, 2016-12-27:
Any idea how to fix this?

tdicola, 2017-02-17:
Ah the indentation was a bit screwy in the example but is fixed now, thanks!

closed 2017-02-17 by tdicola