adafruit/Adafruit-Motor-HAT-Python-Library #19: Android Things support

zugaldia, 2017-01-22:
Hey team, I've started porting this library to Android Things (Java). The [code is here](https://github.com/zugaldia/adafruit-motor-hat) and, though it only supports DC motors, it's already useful for projects:

[image]

Feel free to close this ticket, I I just wanted to let you know about this port. Thanks for opensourcing this Python library :)

ladyada, 2017-01-23:
neat!

caternuson, 2020-10-02:
Closing since this was just a project share. But for anyone coming across this, be aware:
This library is being deprecated. Please update to using the new library here:
https://github.com/adafruit/Adafruit_CircuitPython_MotorKit

closed 2020-10-02 by caternuson