adafruit/Adafruit-Motor-HAT-Python-Library #22: This library doesn't work with Python 3 (and it's 2017)

wroscoe, 2017-02-13:
This should be at least noted on the repo so people waist hours trying. This port should not be too difficult and there are several branches/pull requests that have made it working for their application. 

The [PMW Servo Hat](https://www.adafruit.com/product/2327) has a great install flow that works for python 3. Could this [install guide](https://learn.adafruit.com/adafruit-dc-and-stepper-motor-hat-for-raspberry-pi) be made to match. 

I'm using this to allow differential drive cars use the [donkey](https://github.com/wroscoe/donkey to make them self driving.

wroscoe, 2017-02-13:
Turns out there is an easy fix. 
```
cd Adafruit-Motor-HAT-Python-Library
2to3  -w .
pip install . -I
```

samclane, 2017-02-16:
@wroscoe I tried using the method you described. While the process completed, there are two noticeable problems:
1. The library doesn't work outside of the ../Adafruit_MotorHAT_Python/ directory
2. The example stepper code given in the tutorial fails at the line:

`myStepper.step(100, Adafruit_MotorHAT.FORWARD, Adafruit_MotorHAT.SINGLE)`

giving the error:

`Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/Adafruit-Motor-HAT-Python-Library/Adafruit_MotorHAT/Adafruit_MotorHAT.py", line 158, in step
    lateststep = self.oneStep(direction, stepstyle)
  File "/home/pi/Adafruit-Motor-HAT-Python-Library/Adafruit_MotorHAT/Adafruit_MotorHAT.py", line 135, in oneStep
    coils = step2coils[self.currentstep/(self.MICROSTEPS/2)]
TypeError: list indices must be integers, not float
`
This is due to the way Python3 casts division to a float. 

Is there a way to convert around this using 2to3 too? Or are there few enough calls that I can just manually change them?

wroscoe, 2017-02-16:
@samclane 
1. I was doing this with a virtualenv activated and it does work outside that one directory. 
2. In the readout of 2to3 I remember seeing some unfixable errors in that script. I'm only using the DCMotors script so I don't have any answers for you. I learned about the 2to3 script this week.

tdicola, 2017-02-17:
Ah thanks for the reminder, yeah this library needed some bigger changes because the python-smbus library was never ported to python 3.  Luckily I made my own pure python version of an smbus/i2c interface that supports both py2 and 3.  Just swapped over the lib to use this and fixed a few other python 3 issues (relative imports).  Tested it out on both python 2 and 3 and it works great.  Grab the latest master code and reinstall (I recommend first completely uninstalling the old lib with `sudo pip uninstall adafruit-motorhat` or `sudo pip3 uninstall adafruit-motorhat` for py 3).

closed 2017-02-17 by tdicola

woody1, 2018-02-14:
Hi 

I am running the examples with Python 2 and it works a treat :)  However, when I try python3 I am getting the following error:

pi@raspberrypi:~/Adafruit-Motor-HAT-Python-Library/examples $ sudo python3 StepperTest.py

Traceback (most recent call last):
  File "StepperTest.py", line 3, in <module>
    from Adafruit_MotorHAT import Adafruit_MotorHAT, Adafruit_DCMotor, Adafruit_StepperMotor
ImportError: No module named 'Adafruit_MotorHAT'

Would you be able to give me some guidance on how to run this with Pythin 3.

Thanks 

Woody

woody1, 2018-02-14:
Ref my last post - 
I installed it using sudo python setup.py install
and not sudo python3 setup.py install
Now works a treat - Thanks