adafruit/Adafruit-Motor-HAT-Python-Library #24: Different torque for a stepper motor with microstepping

Alfred17, 2017-10-11:
Raspberry Pi3 with Raspian
python 2.7.9

In one direction of rotation, the library switches off the current at maximum. Furthermore, the coils are switched one step too early.
I solved the problem like this:
 
        if (style == Adafruit_MotorHAT.MICROSTEP):
            self.currentstep -= 1

            # go to next 'step' and wrap around
            self.currentstep += self.MICROSTEPS * 4
            self.currentstep %= self.MICROSTEPS * 4

            pwm_a = wm_b = 0
            if (self.currentstep >= 0) and (self.currentstep < self.MICROSTEPS):
                pwm_a = self.MICROSTEP_CURVE[self.MICROSTEPS - self.currentstep]
                pwm_b = self.MICROSTEP_CURVE[self.currentstep]
            elif (self.currentstep >= self.MICROSTEPS) and (self.currentstep < self.MICROSTEPS*2):
                pwm_a = self.MICROSTEP_CURVE[self.currentstep - self.MICROSTEPS]
                pwm_b = self.MICROSTEP_CURVE[self.MICROSTEPS*2 - self.currentstep]
            elif (self.currentstep >= self.MICROSTEPS*2) and (self.currentstep < self.MICROSTEPS*3):
                pwm_a = self.MICROSTEP_CURVE[self.MICROSTEPS*3 - self.currentstep]
                pwm_b = self.MICROSTEP_CURVE[self.currentstep - self.MICROSTEPS*2]
            elif (self.currentstep >= self.MICROSTEPS*3) and (self.currentstep < self.MICROSTEPS*4):
                pwm_a = self.MICROSTEP_CURVE[self.currentstep - self.MICROSTEPS*3]
                pwm_b = self.MICROSTEP_CURVE[self.MICROSTEPS*4 - self.currentstep]

        # only really used for microstepping, otherwise always on!
        self.MC._pwm.setPWM(self.PWMA, 0, pwm_a*16)
        self.MC._pwm.setPWM(self.PWMB, 0, pwm_b*16)

        # set up coil energizing!
        coils = [0, 0, 0, 0]

        if (style == Adafruit_MotorHAT.MICROSTEP):
            if (self.currentstep >= 0) and (self.currentstep < self.MICROSTEPS):
                if (dir == Adafruit_MotorHAT.FORWARD):
                    coils = [1, 1, 0, 0]
                else:
                    coils = [1, 0, 0, 1]
            elif (self.currentstep >= self.MICROSTEPS) and (self.currentstep < self.MICROSTEPS*2):
                if (dir == Adafruit_MotorHAT.FORWARD):
                    coils = [0, 1, 1, 0]
                else:
                    coils = [0, 0, 1, 1]
            elif (self.currentstep >= self.MICROSTEPS*2) and (self.currentstep < self.MICROSTEPS*3):
                if (dir == Adafruit_MotorHAT.FORWARD):
                    coils = [0, 0, 1, 1]
                else:
                    coils = [0, 1, 1, 0]
            elif (self.currentstep >= self.MICROSTEPS*3) and (self.currentstep < self.MICROSTEPS*4):
                if (dir == Adafruit_MotorHAT.FORWARD):
                    coils = [1, 0, 0, 1]
                else:
                    coils = [1, 1, 0, 0]
        else:
            step2coils = [     [1, 0, 0, 0],
                [1, 1, 0, 0],
                [0, 1, 0, 0],
                [0, 1, 1, 0],
                [0, 0, 1, 0],
                [0, 0, 1, 1],
                [0, 0, 0, 1],
                [1, 0, 0, 1] ]
            coils = step2coils[self.currentstep//(self.MICROSTEPS//2)]

        #print "coils state = " + str(coils)
        self.MC.setPin(self.AIN2, coils[0])
        self.MC.setPin(self.BIN1, coils[1])
        self.MC.setPin(self.AIN1, coils[2])
        self.MC.setPin(self.BIN2, coils[3])

        return self.currentstep

Alfred17, 2017-10-12:
Posted too fast, the direction of rotation was wrong, so it's right. It must be called BACKWARD instead of FORWARD in the query:

        if (style == Adafruit_MotorHAT.MICROSTEP):
            if (self.currentstep >= 0) and (self.currentstep < self.MICROSTEPS):
                if (dir == Adafruit_MotorHAT.BACKWARD):
                    coils = [1, 1, 0, 0]
                else:
                    coils = [1, 0, 0, 1]
            elif (self.currentstep >= self.MICROSTEPS) and (self.currentstep < self.MICROSTEPS*2):
                if (dir == Adafruit_MotorHAT.BACKWARD):
                    coils = [0, 1, 1, 0]
                else:
                    coils = [0, 0, 1, 1]
            elif (self.currentstep >= self.MICROSTEPS*2) and (self.currentstep < self.MICROSTEPS*3):
                if (dir == Adafruit_MotorHAT.BACKWARD):
                    coils = [0, 0, 1, 1]
                else:
                    coils = [0, 1, 1, 0]
            elif (self.currentstep >= self.MICROSTEPS*3) and (self.currentstep < self.MICROSTEPS*4):
                if (dir == Adafruit_MotorHAT.BACKWARD):
                    coils = [1, 0, 0, 1]
                else:
                    coils = [1, 1, 0, 0]
        else:

caternuson, 2020-10-02:
This library is being deprecated. Please update to using the new library here:
https://github.com/adafruit/Adafruit_CircuitPython_MotorKit
If you continue to have this issue, please open a new issue in that repo.

closed 2020-10-02 by caternuson