adafruit/Adafruit-Motor-HAT-Python-Library #27: DC Motor Control can't go all on

MZachmann, 2018-08-16:
Hi,

I might be wrong, but...

There's sort of a fundamental misunderstanding as to how the PWM chip converts on/off values to waves. It takes two counter values that range from 0...4095. So you can not set a pin value to 4096. In binary 12 bits that's a zero. If you want to set it all on the value needs to be set to 4095 - which is all ones in binary.

So, the setSpeed method can never go fully on or off (because 255*16 is not 4095) and the on/off methods don't do what you think they're doing since 4096 and 0 are the same values. Calling setPWM(pin, 0, 4096) is just gibberish and should be (pin, 0, 4095).

I can submit a fix if you'd like. I've changed my code so setSpeed takes 0...4095.

ladyada, 2018-08-16:
hiya, can you point out the line you're referring to, that way we can be sure we're discussing the same thing :)

MZachmann, 2018-08-16:
Yeah, the setSpeed method takes 0...255 but that makes the maximum value 255*16 so a speed of 255 isn't full on. It was trivial to fix but I did have to notice it. See line 213: self.MC._pwm.setPWM(self.PWMpin, 0, speed*16).

ladyada, 2018-08-17:
we're actually about to replace this library completely...can you wait a few days? :)

MZachmann, 2018-08-17:
Honestly, I'm good. It was easy to fix on my end. I just wanted to inform you folks. Take your time.

caternuson, 2020-10-02:
This library is being deprecated. Please update to using the new library here:
https://github.com/adafruit/Adafruit_CircuitPython_MotorKit
If you continue to have this issue, please open a new issue in that repo.

closed 2020-10-02 by caternuson