adafruit/Adafruit-Motor-HAT-Python-Library #29: Motor state should be stored as part of the class and retrievable from calling code

Hobadee, 2018-11-25:
# Problem:
Currently there is no way of ascertaining what a motor is doing; it must be tracked manually in the users code.  This may create issues when multi-threading or in complex functions where manually tracking the status of the motors is undesirable.

The `Adafruit_StepperMotor` and `Adafruit_DCMotor` classes should store this information locally and be able to return it when requested.

# Solution
To implement this, I propose the following new methods:
In class `Adafruit_StepperMotor`:
* `getSpeed()`
  * Takes no arguments
  * returns RPM

(I'm not familiar with running stepper motors, so other GET methods as necessary.  This was the only one I could tell was necessary from looking at the source)

In class `Adafruit_DCMotor`
* `getSpeed()`
  * Takes no arguments
  * Returns speed (0-255)
* `getCommand()`
  * Takes no arguments
  * Returns `Adafruit_MotorHAT` ENUM
    * Adafruit_MotorHAT.FORWARD
    * Adafruit_MotorHAT.BACKWARD
    * Adafruit_MotorHAT.BRAKE
    * Adafruit_MotorHAT.RELEASE

Additionally, some class variables would have to be added to track this information.  Class methods would have to be updated to use these variables instead of method variables they currently use.

-----

I would do this myself and submit a pull request, but I'm new to Python and this code is still slightly over my head.

bunchc, 2018-12-14:
I think this can be done by querying the hardware rather than storing the state.
**Note:** This is all very rough, and does not account for speed or stepper motors just yet.

First, I needed to add a `getPWM` method to the `PWM` class.
```python
    def getPWM(self, channel):
        "Gets a single PWM channel"
        register = {
            '__LED0_ON_L': self.i2c.read8(self.__LED0_ON_L+4*channel),
            '__LED0_ON_H': self.i2c.read8(self.__LED0_ON_H+4*channel),
            '__LED0_OFF_L': self.i2c.read8(self.__LED0_OFF_L+4*channel),
            '__LED0_OFF_H': self.i2c.read8(self.__LED0_OFF_H+4*channel)
        }
        return register
```

With that, we can work out what the various commands look like and create our `getCommand` method in the `Adafruit_DCMotor` class:

```python
    def getCommand(self):
        IN1pinState = self.MC._pwm.getPWN(self.IN1pin)
        IN2pinState = self.MC._pwm.getPWN(self.IN2pin)
        if (IN1pinState['__LED0_ON_H'] == 16 and IN2pinState['__LED0_ON_H'] == 0):
            return Adafruit_MotorHAT.FORWARD
        elif (IN1pinState['__LED0_ON_H'] == 0 and IN2pinState['__LED0_ON_H'] == 16):
            return Adafruit_MotorHAT.BACKWARD
        elif (IN1pinState['__LED0_ON_H'] == 0 and IN2pinState['__LED0_ON_H'] == 0):
            return Adafruit_MotorHAT.RELEASE
```

I'm happy to clean up my fork and send a pull request if there is interest.

ladyada, 2018-12-14:
hiya we're migrating away from this library and to MotorKit - can you take a look there? we'll deprecate this library and supporrt only this one soon (the instructions have been updated):
https://github.com/adafruit/Adafruit_CircuitPython_MotorKit

bunchc, 2018-12-14:
@ladyada Thanks for the heads up, looks like this will do what is needed.

caternuson, 2020-10-02:
Closing. Seems like the new library fixes this:
https://github.com/adafruit/Adafruit_CircuitPython_MotorKit

closed 2020-10-02 by caternuson