adafruit/Adafruit-Motor-HAT-Python-Library #31: scripts/drive.py is not working.please help me,,creating donkey based differential drive

jishnuj741, 2019-03-06:
python scripts/drive.py --remote http://163.v.v.x:8887
Detected running on rasberrypi. Only importing select modules.
Using TensorFlow backend.
Traceback (most recent call last):
  File "scripts/drive.py", line 35, in <module>
    mythrottlecontroller = dk.actuators.PCA9685_Controller(cfg['throttle_actuator_channel'])
AttributeError: 'module' object has no attribute 'PCA9685_Controller'

ladyada, 2019-03-06:
that isn't using our code? please contact the vendor you got the controller from :)

closed 2019-03-06 by ladyada

jishnuj741, 2019-03-07:
@ladyada bought adafruit hat and i can access using serve .py in my laptop but doubt which code should i burn in pi usng virtual env.is it diff_drive.py or drive.py?

jishnuj741, 2019-03-07:
@ladyada python diff_drive.py
Detected running on rasberrypi. Only importing select modules.
Using TensorFlow backend.
diff_vehicle
{'steering_actuator_channel': None, 'vehicle_id': 'mycar', 'throttle_actuator_channel': None, 'pilot_model_path': '/home/pi/mydonkey/models/default.h5', 'throttle_actuator_min_pulse': None, 'vehicle_loop_delay': 0.08, 'camera_loop_delay': 0.08, 'steering_actuator_min_pulse': None, 'throttle_actuator_zero_pulse': None, 'steering_actuator_max_pulse': None, 'throttle_actuator_max_pulse': None}
Traceback (most recent call last):
  File "diff_drive.py", line 31, in <module>
    left_motor = dk.actuators.Adafruit_Motor_Hat_Controller(cfg['left_actuator_channel'])
KeyError: 'left_actuator_channel'

defined the key in the file

ladyada, 2019-03-10:
can you send a link to the product you're using?