adafruit/Adafruit-Motor-HAT-Python-Library #32: Adafruit hat differential drive help

jishnuj741, 2019-03-07:
python diff_drive.py
Detected running on rasberrypi. Only importing select modules.
Using TensorFlow backend.
diff_vehicle
{'steering_actuator_channel': None, 'vehicle_id': 'mycar', 'throttle_actuator_channel': None, 'pilot_model_path': '/home/pi/mydonkey/models/default.h5', 'throttle_actuator_min_pulse': None, 'vehicle_loop_delay': 0.08, 'camera_loop_delay': 0.08, 'steering_actuator_min_pulse': None, 'throttle_actuator_zero_pulse': None, 'steering_actuator_max_pulse': None, 'throttle_actuator_max_pulse': None}
Traceback (most recent call last):
  File "diff_drive.py", line 31, in <module>
    left_motor = dk.actuators.Adafruit_Motor_Hat_Controller(cfg['left_actuator_channel'])
KeyError: 'left_actuator_channel'

defined the key in the file,but not accessing the left_actuator_channel key... please help me.. should i include anything

ladyada, 2019-03-09:
hiya diff_drive.py isn't code in our repo, and neither is Adafruit_Motor_Hat_Controller 

are you using some other code because I dont think its this one! 

closing because i think its some other code thats the issue :)

closed 2019-03-09 by ladyada

jishnuj741, 2019-03-11:
run the code python scripts/drive.py --remote http://ip address of server
Traceback (most recent call last):
  File "scripts/drive.py", line 57, in <module>
    right_motor =dk.actuators.Adafruit_Motor_Hat_Controller(cfg['right_actuator_channel'])
KeyError: 'right_actuator_channel'
this is the script i have run..is this the